kidoman/embd #24: SPI not working of RPI2

fredcarle, 2015-03-08:
There is a problem with trying to acquire data over SPI on RPI2. I get an invalid argument error and I think it has to do with the BCM2835 base. Is there a solution around this?

kunalpowar, 2015-03-30:
Thanks for sharing this @fredcarle. We've not really tested this on RPI2. We'll surely see what can be done about this. As you already have a RPI2 at your disposal, you send a PR with the fix.

hbhasker, 2015-09-03:
fred, I ran into a similar issue with my raspberry pi b rev2. The problem is the SPIIOCTransfer struct is missing a couple of fields which I think cause the ioctl calls to fail. Once I added the missing fields in my code works fine. Maybe something is similar with rpi2.

Check details on https://github.com/kidoman/embd/issues/38

tve, 2016-09-25:
This is fixed by #69

closed 2016-09-25 by tve