kidoman/embd #29: LCD Driver

aleksclark, 2015-05-22:
Sup! I just put up an LCD driver that leverages embd, was wondering if you thought it should be merged into the main codebase. Since I wanted to be able to update the LCDs from different goroutines, I also implemented a shared i2c writer and port abstraction:

https://github.com/aleksclark/portwriter
https://github.com/aleksclark/embd-lcd
https://github.com/aleksclark/sharedi2c

closed 2015-05-23 by aleksclark

kidoman, 2015-05-24:
Why did you close it?

aleksclark, 2015-05-24:
Hey Karan,

I only just noticed that there was already an i2c/lcd interface already
included in embd, while browsing the embd forks. It seems to work in a
straightforward manner :) Right now the only feature that's really not
there is the channel-based i2c writer, I need to think about how to go
about implementing that with the existing code, or if it's even worth doing
so.