kidoman/embd #31: not compatible with RPI2

rodcar125, 2015-06-11:
Hi!
Was getting nothing from the server on chilipepper until I searched my log file for the server and found this...

2015/06/10 23:58:40 hub.go:95: Error: embd: your host ": ARMv7 Processor rev 5 (v7l)" is not supported at this moment. request support at https://github.com/kidoman/embd/issues

So im here trying to get support and if theres anything I can do to help please let me know although my programming skills are basically non existent, lol. Is there a plan to release a version that supports RPI2?

kidoman, 2015-06-11:
Actually we had tested the latest release with RPi 2. Are you using the latest embd version?

rodcar125, 2015-06-18:
Im sorry, i got confused i believe, its working. I dont know what i did wrong.

I have the embd version 0.1.0 is that the latest one?

closed 2016-09-25 by tve