kidoman/embd #32: Support GPIO pull-ups

captncraig, 2015-06-29:
The pull-up and down functionality is currently unimplemented in the generic host. I would really like this functionality to be available for rpi. 

My current workaround is using github.com/stianeikeland/go-rpio only to set the pull-ups. I'd really rather stick to embd only because I need the i2c and other awesome features.

Their implementation is at https://github.com/stianeikeland/go-rpio/blob/master/rpio.go#L232. It involves some tricky memory mapping of a register file that I am not super comfortable with.

kidoman, 2015-06-29:
I will give it a shot :)

snikch, 2016-04-09:
Hey @kidoman  - did you ever get a chance to have a look up pull up/down?

ivoras, 2017-09-29:
Just as a notice to anyone in 2017: pullups are still not implemented.
Basically this library cannot be used for driving buttons.

closed 2025-06-14 by captncraig