kidoman/embd #34: PWM on RPi B+ not working

akofoed, 2015-07-11:
Thanks for the cool EMBD framework. It's really great.

I see in the docs for the PI B+ that the GPIO18 should work as PWM pin. However it's not supported in the host (host/rpi/rpi.go).

I get this if I try:
> Error in NewPWMPin (18)!
> panic: gpio: pwm not supported on this host

Code:
```
	pinNo := 18
	pwm, err := embd.NewPWMPin(pinNo)
	if err != nil {
		fmt.Println("Error in NewPWMPin (18)!")
		panic(err)
	}
	defer pwm.Close()
	// 500 Hz == 2.000.000 ns
	pwm.SetDuty(2000000)
```

closed 2016-08-04 by akofoed

chaosaffe, 2016-11-22:
@akofoed did you resolve this, or are the doc's incorrect on supporting PWM for the Pi?

akofoed, 2016-11-22:
Hi chaosaffe
No, I wrote it in C, so problem solved.