kidoman/embd #43: Looking for a maintainer

kidoman, 2015-12-26:
Hello all,

It sucks when you gave your heart and soul to some project but it is falling into disrepair because of lack of time. Is anyone willing to help me manage the project, and take EMBD forward. Looking for some input here.

Regards,
Karan Misra

jtwatson, 2016-03-20:
I would be very interested in helping to take this project forward.  Please let me know what you have in mind.

- Joseph Watson

Carrotman42, 2016-04-14:
I think this project is awesome. Sadly I haven't really found a use for it yet, but I hope one day I can find the time to use it and that the project will still be alive and well for me to use it.

I would definitely be able to contribute by doing code reviews on pull requests.

viru, 2016-06-21:
Hi,

I'm using embd in one project (simple robot based on rpi2) and I'm happy to help at least with rpi testing and maintaining issues / reviewing PRs.

Let me know how can I help.

tve, 2016-09-06:
I've added support for CHIP to my fork (https://github.com/tve/embd) and also merged all PRs I could make sense of. Looks like I'm going to be using this library myself, so I'll be maintaining for a while.

lukepalmer, 2017-05-07:
Happy to help. We have a few key PRs that are needed to keep things going, like current Raspberry Pi 3 support. Let me know.

kidoman, 2017-05-07:
Thanks for offering to help. I would like to be able to revitalize this project. Adding you to the project.