kidoman/embd #45: proposal for a better i2c interface / sensors

quinte17, 2015-12-30:
hey guys.

i like the idea of embd beeing a plattform for hardware guys ;)
but as much as i like the idea, when it comes to i2c i somehow dislike the interface used.
in my opinion https://github.com/davecheney/i2c has a much cleaner interface.
if you look at the example i implemented:
https://github.com/quinte17/bme280/blob/master/bme280.go
you might see what i mean.

* sensors should not have the address hardcoded (bmp085). some sensors also support other addresses. for example the bme280 supports address 0x77 or 0x76. it depends on hw-configuration. the bus should handle the addressing.
* many sensor registers are lined up. so you could read them all in one go.
* use encoding/binary to transfer from []byte to a complicated struct

plz dont missunderstand this proposal. i just searched for an interface i could implement, so the sensor could be used in other places too. atm i only found embd as the only project. i believe my implementation doesnt fit well for embd.
just want to start some discussion...

yours

tve, 2016-09-06:
I don't understand "many sensor registers are lined up. so you could read them all in one go." in that you can read multiple sensor registers using:
```
// ReadFromReg reads n (len(value)) bytes from the given address and register.
func ReadFromReg(addr, reg byte, value []byte) error
```

quinte17, 2016-09-06:
sry, my fault, I missed that one. havn't looked at all functions.

another question i have:
are there any plans to migrate to
https://github.com/golang/exp/tree/master/io/
?

tve, 2016-09-09:
I didn't know about that. What would the benefit be? It's not really clear to me who maintains that and what the plans are...

quinte17, 2016-09-09:
Here is the thread which initially started this project.
https://groups.google.com/d/msg/golang-dev/ofaaIJPWRKg/5QuiOPlZBgAJ