kidoman/embd #48: Errors using the syscall package when importing github.com/kidoman/embd/host/rpi

danhardman, 2016-01-14:
There's a bunch of errors when I import `github.com/kidoman/embd/host/rpi` saying:
`undefined: syscall...` I think this is to do with the note mentioned in the overview of: https://golang.org/pkg/syscall/

> NOTE: This package is locked down. Code outside the standard Go repository should be migrated to use the corresponding package in the golang.org/x/sys repository. That is also where updates required by new systems or versions should be applied. See https://golang.org/s/go1.4-syscall for more information.

I'm developing on Windows 10 and using go version go1.5.3 windows/amd64

kidoman, 2016-01-18:
Hi,

Did it start working?

Regards,
Karan Misra

danhardman, 2016-01-20:
I was being an idiot. I was developing in atom using the go-plus package which builds files on save. I didn't have GOOS=linux set so when it got to building files with the `syscall` package imported, it couldn't find the linux variables (because I'm on windows).

Moral of the story: If you want to use anything that uses the `syscall` package on anything other than linux, set GOOS=linux.

closed 2016-09-25 by tve