kidoman/embd #57: go get error

mattetti, 2016-05-09:
Probably not a big deal but wanted to mention it

```
$ go get -u github.com/kidoman/embd/...
# github.com/kidoman/embd/host/generic
../../../github.com/kidoman/embd/host/generic/interrupt.go:50: undefined: syscall.EpollCreate1
../../../github.com/kidoman/embd/host/generic/interrupt.go:57: undefined: syscall.EpollEvent
../../../github.com/kidoman/embd/host/generic/interrupt.go:60: undefined: syscall.EpollWait
```

kidoman, 2016-05-09:
Thanks for reporting this... will have a look

mattetti, 2016-05-09:
np, this issue seems to be due to the fact that the syscall doesn't exist on OS X so if one were to pull the package on that os, the error would occur. You might need to split the file into multiple files and use build tags.

declanshanaghy, 2016-11-30:
From as far as im seeing there's no option but to pull the host/generic package because host/rpi imports. 

How does the sample in the video shown here compile on OSX when the host/rpi package is imported? - http://embd.kidoman.io/

pravipati, 2017-12-07:
@kidoman I was wondering if you've had a chance to look at this issue. Per @declanshanaghy's comment ☝️, I'd be curious to see how you were able to compile on OSX in the demo video. For now, I'm working on splitting my package up (one version for OS X and one for rpi).

ripienaar, 2018-01-05:
`GOOS=linux GOARCH=arm GOARM=5 go build` use a dependency manager like deb or glide to get the code on the build env

pwaterz, 2018-05-05:
Another super awesome tool is https://github.com/goreleaser/goreleaser