kidoman/embd #66: is this project abandoned?

tve, 2016-08-27:
Looks like lots of good pull requests and valid issues but no movement for a long time...

I created a fork at https://github.com/tve/embd and started to pull in the various PRs I can grok, plus am adding support for NextThing's CHIP. If the original author has lost interest, maybe there are other folks that are interested to continue? I don't want to step on anyone's toes here, all I'd like is for this nice library to continue on.

kidoman, 2016-09-07:
Hi `tve` Can we talk? Thanks for talking initiative and I would like to come up with a way to take this lib ahead. I got too busy in my startup for the last couple of years and could not give it the time it deserved.

tve, 2016-09-07:
Sure: tve at voneicken dotcom or the embd gophers slack channel or @ tve in the gopgers slack

fhedberg, 2016-09-24:
Did anything come out of this discussion?

tve, 2016-09-24:
https://github.com/kidoman/embd/pull/69 is one outcome
Right now I am focusing on https://groups.google.com/forum/#!topic/golang-dev/ofaaIJPWRKg, see the tail of the thread
Which platform are you interested in?

fd0, 2017-03-09:
I'm also willing to help out

fd0, 2017-04-22:
FYI: I'm not interested any more, I've switched to https://periph.io/. Thanks for your work!