kidoman/embd #82: pi zero w support

olliephillips, 2017-03-16:
Noted that unless I build with `GOARM=6` the compiled binary will not run on pi zero w, it outputs "Illegal Instruction"  Happy to amend README, since I think other newcomers might find the same, and be deterred.

I have `LED0` blinking,  but should I expect embd library to work on pi zero w with existing Rasp Pi board definition files?

danesparza, 2017-04-07:
Hmmm ... what version of Go are you using?  With Go 1.8, I just needed to `export GOOS=linux` and
  `export GOARCH=arm` in order to build the correct arm binary on Windows (that I then SCP'd to my Raspberry Pi Zero).

olliephillips, 2017-04-07:
I'm cross compiling from a Mac on Go 1.8

It's like it is using ```GOARM=7``` by default, since if I use 7 I also get same error as if GOARM env var omitted.

```
pi@pizero3:~ $ sudo ./simpleblinker
Illegal instruction
```

So I need to explicitly set ```GOARM=6``` for it to work.