kidoman/embd #88: Invalid host

fallais, 2017-12-19:
Hello,

I have this error : `host: invalid host \"Raspberry Pi\"`
What can I do please ?
I use your library in a Golang program that is running in a Docker container in a Raspberry 3.

Thanks a lot.

Carrotman42, 2017-12-26:
An example piece of code and instructions to reproduce would be very useful
in helping you both.

marian-craciunescu, 2018-01-22:
I'm having the same issue after the fork.The imports paths when forking are still `github/kidoman/embd` 
and you will have an empty  `describers` map.

vacoj, 2018-02-07:
Same issue here.  +1

dougthebug, 2018-03-16:
Same here. I downgraded to an earlier kernel to get around this, qmsk/e2 on rpi2.  I just posted there my solution, but I’d rather try to wrap my head around how this works so an update doesn’t lead to my stuff breaking.  😭

philippegabert, 2018-03-17:
I think you guys should add this to the import list... It solved the issue for me

`_ "github.com/kidoman/embd/host/rpi"`